Obijuan/digital-electronics-with-open-FPGAs-tutorial #11: Problema con los elementos "memoria" del tutorial 19

fernandobrea, 2018-11-01:
Hola.
Los elementos memoria de los ejercicios del tutorial 19, no tienen definido el campo "formato", no generando ningún problema porque en el .ice no aparece el campo format.
El problema surge cuando se cambia el nombre de estos elementos "memoria" sin "añadir" el formato, que está en blanco. 

[image]

A partir de ese momento si se guarda el circuito, en el .ice se incorpora el campo format a null, 

[image]

Esto provoca que a la  hora de abrir el archivo, cuando llega a este elemento no carga su información y no termina de cargar nunca. No provoca ningún mensaje de error.

[image].

Solamente ocurre con estos elementos memoria de los ejercicios, pues al añadir uno nuevo, si o si hay que rellenar el campo format.
También he verificado que ese es el problema, pues he modificado el .ice cambiando null por 10 y el archivo se ha cargado normalmente.
Mi configuración es la siguiente:

Sistema Operativo: Windows 10.
Version Icestudio: 0.3.3
Version Apio: 0.3.6
Coleccion en Icestudio: Academia-Jedi-HW-19
FPGA: Alhambra II

Si necesitáis mas información, no dudéis en pedirla.
Slds.
Fernando.

fernandobrea, 2018-11-03:
Entiendo que la diferencia está en que en la versión de icestudio 0.3.3-beta, este campo format no estaba contemplado, mientras que en la 0.3.3 sí.
Por lo que solamente ocurre al copiar y manipular elementos  "memory" creados con la 0.3.3-beta y salvarlos en la 0.3.3, sin tener la precaución de seleccionar un formato, ya que icestudio permite no seleccionarlo.
Slds.
Fernando.

Obijuan, 2018-11-21:
¡Muchas gracias! Bug corregido! Efectivamente, los ejemplos se hicieron con una versión anterior de icestudio. Luego Jesús añadió el nuevo campo con el formato.  Lo que he hecho ha sido grabar los ejemplos desde icestudio 0.3.3 con los campos a decimal, y he vuelto a crear la colección

+5 Bitpoints por el informe!  ;-)

¡Gracias!

Obijuan, 2018-11-21:
Total Bitpoints: 260/260 (+30)

closed 2018-11-21 by Obijuan

fernandobrea, 2018-11-21:
Muchas gracias a ti por todo tu trabajo.

Fernando.