Obijuan/digital-electronics-with-open-FPGAs-tutorial #17: El archivo cad.list no existe

R-Hidalgo, 2020-06-11:
Estimados,

Al sintetizar sale error "El archivo cad.list no existe" . El proyecto y el archivo se encuentran en la misma carpeta y no hay espacios en los nombres de carpeta.

Utilizo Windows 10, con Icestudio 0.4.0 y Alhambra II

[image]

[image]

[image]

Obijuan, 2020-06-30:
Hola,

Perdona que haya tardado tanto en responder (estoy muy desbordado). Si escribes en el [grupo de FPGAwars]([https://groups.google.com/g/fpga-wars-explorando-el-lado-libre](https://groups.google.com/g/fpga-wars-explorando-el-lado-libre)) la gente te responderá más rápido

Te envío un ejemplo que lo acabo de probar, y funciona bien, para descubrir qué es lo que te está pasando. Es un archivo .zip con el fichero test.list y un ejemplo que lee ese fichero

[test.zip](https://github.com/Obijuan/digital-electronics-with-open-FPGAs-tutorial/files/4849881/test.zip)

Descomprímelo, abre el fichero test-file.ice en icestudio y pulsa en Herramientas/Sintetizar

Te debería salir esto:

[image]

Aunque esta característica de leer ficheos funciona desde las primeras versiones, prefiero que te actualices a la última versión de icestudio (icestudio-nightly-n200523) para estar en las mismas condiciones y eliminar variables de error

Pruébalo y dime el resultado

c0smm0, 2020-06-30:
Hola Juan,
Me temo, que he recibido este correo por error, pero aprovecho para saludarte y darte las gracias por tu magnifica contribución a la humanidad.

Mariano Moreno Pérez
Inplatec Plásticos, S.L.

[cid:a7b97da6-b948-438e-90cf-72f356dddee2]916420333

[cid:76501b0e-c223-4cd1-8f3d-fcaf63937f91]
http://inplatec.com/
P Antes de imprimir, por favor considera el medio ambiente.

Por favor, piense en el medio ambiente antes de imprimir este mensaje.

Si usted recibe por error este mensaje, por favor comuníquelo a su r

emitente y borre inmediatamente tanto el mensaje como cualquier anexo

o copia del mismo, ya que contiene información confidencial, dirigida

exclusivamente a su destinatario y cuya utilización o divulgación a terceros

están prohibidas por la ley, pudiendo dar lugar a responsabilidades civiles y/o

penales.

Esta comunicación contiene información que es confidencial y también puede contener información privilegiada.

Es para uso exclusivo del destinatario/s. Si usted no es el destinatario/s tenga en cuenta que cualquier distribución,

copia o uso de esta comunicación o la información que contiene está estrictamente prohibida.

Si usted ha recibido esta comunicación por error por favor notifíquelo por correo electrónico o por teléfono.

En cumplimiento de lo dispuesto en el Reglamento 2016/679, del Parlamento Europeo y del Consejo, de 27 de abril de 2016,

Inplatec Plasticos SL pone en su conocimiento que esta información ha sido remitida por personal al servicio de la citada

empresa con la finalidad de cumplimiento de las funciones de su competencia. Ponemos en su conocimiento la posibilidad

de ejercer sus derechos de acceso, rectificación, supresión, oposición, portabilidad de datos, limitación del tratamiento y

olvido en los términos establecidos en la legislación vigente, que podrá hacer efectivos dirigiéndose por escrito a

Inplatec plasticos SL, con dirección en Calle León nº9, Polígono Industrial Cobo Calleja 28947, Fuenlabrada (Madrid).

Si desea dejar de recibir correos de esta lista, reenvíe este e-mail a ncaballero@inplatec.com<mailto:ncaballero@inplatec.com>

[image]<https://user-images.githubusercontent.com/1286062/86092646-16ee1900-baae-11ea-903d-ad4bd7eeb733.png>

Obijuan, 2020-06-30:
Hola Mariano,

Lo que te ha llegado es la notifiación de github porque estás suscrito a este proyecto. Todos los mensajes los puedes ver en esta URL:

https://github.com/Obijuan/digital-electronics-with-open-FPGAs-tutorial/issues/17

Para no recibir maś notificaciones pincha en el botón unwatch de la parte superior de esa página

R-Hidalgo, 2020-07-01:
Gracias Obijuan,

Al descomprimir los archivos recibidos se ve el problema. Mi archivo de texto no tenía la extensión correcta. Una vez editado el nombre directamente en la carpeta, el proyecto se ejecuta bien.

[image]

closed 2020-07-01 by R-Hidalgo

Obijuan, 2020-07-01:
Great! :smile: